arduino-libraries/USBHost #14: [DOC] https://www.arduino.cc/en/Reference/USBHost only talks about Due...

q2dg, 2016-07-27:
...but Zero/MKR1000 have USBHost capabilities, haven't they?

MisterAwesome23, 2020-03-23:
Hey :)
I see this is still an issue since the site mentioned mentions only Due and not Zero or MKR1000.
Should I take this issue, by confirming these two do have the capability or not and then updating the reference doc as well?

per1234, 2020-03-23:
@MisterAwesome23 the content of https://www.arduino.cc/en/Reference/USBHost is not hosted in a public repository so there is no way for you to update that content.

If you like, you could post the exact text you recommend adding/changing in that page. If done right, this could make it a very quick copy/paste job for the people who do have the access to edit this content.

MisterAwesome23, 2020-03-23:
Oh got you. I thought even those pages were accessible for pushing changes. Anyway, should i confirm the capability of other two for USBHost and draft exact changes to respective authority?

Also, hello again @per1234. Are you something special for the arduino official libraries (cause i see you everywhere 😁), asking because I'm a GSoC20 applicant and you might guide me with some starter issues, since my project aims adding and improving official library examples.

per1234, 2020-03-23:
>I thought even those pages were accessible for pushing changes.

Of the reference pages, only the Arduino Language documentation is available for contributions via pull requests:
https://github.com/arduino/reference-en
The library reference pages are not in a public repository.

>should i confirm the capability of other two for USBHost and draft exact changes to respective authority?

I think it would be helpful, but I'm not a member of Arduino's documentation team so I don't have any say in whether your suggestion will be implemented or not.

>Are you something special for the arduino official libraries

Well, "special" would probably not be an accurate description, but I do help out with maintaining Arduino's GitHub repositories. I normally spend more time with the issue trackers than reviewing pull requests, but I'm one of Arduino GSoC administrators so I'm trying to help out with the large number of pull requests we're getting from the GSoC students right now.

>I'm a GSoC20 applicant and you might guide me with some starter issues, since my project aims adding and improving official library examples.

Let's keep the discussion of your GSoC proposal in your [dedicated issue](https://github.com/arduino/summer-of-code/issues/110) in the arduino/summer-of-code repository. That way, everyone from Arduino who is involved with the program will be able to participate in the conversation.

MisterAwesome23, 2020-03-23:
Oh sure :)
I'll keep GSoC on my dedicated issue tracker. Hope to see you more and fix and help in more things along side :)
Period.