arduino-libraries/USBHost #15: Numeric keypad support ?

andi968, 2017-03-29:
Is it possible to add numeric keypad support to the library?
The /*-+ all work fine, however the 0-9 do not but do produce the following response 
Actual Key pressed          GetOemKey response 
1                                         = 89,
2                                         = 90,
3                                         = 91,
4                                         = 92,
5                                         = 93,
6                                         = 94,
7                                         = 95,
8                                         = 96,
9                                         = 97,
0                                         = 98

Pigeo, 2017-07-20:
Be careful : you have to take into account the status of the numlock in order to know e.g. if key 92 should be interpreted as "4" or as "left arrow"... It would be nice if the library could handle that for us.

Pigeo, 2017-07-20:
I have just tested on my DUE and to me there's no bug : this library is working as expected (by default, the numlock is off and the keypad keys don't give numbers, but if you first press the numlock, then you'll have numbers).

What is maybe confusing you is that the numlock is off by default. Would be great to have a function to change the status of numlock by software, or to put it ON by default, but I don't know if the HID keyboard boot protocol allows us to do that.